STMicroelectronics/STMems_Android_Sensor_HAL_IIO #4: Compile fatal error: iostream: No such file or directory

Tiny-Angel305, 2019-08-28:
Integrated this sensor with qcom snapdragon (dart-sd410c) source code, and tried to build full source code.
While compiling, got this error.
target thumb C++: sensors.msm8916_32 <= hardware/STMicroelectronics/SensorHAL_IIO/src/SensorHAL.cpp
  CHK     include/generated/uapi/linux/version.h
target thumb C++: sensors.msm8916_32 <= hardware/STMicroelectronics/SensorHAL_IIO/src/utils.cpp
target thumb C++: sensors.msm8916_32 <= hardware/STMicroelectronics/SensorHAL_IIO/src/CircularBuffer.cpp
hardware/STMicroelectronics/SensorHAL_IIO/src/utils.cpp:10:20: fatal error: iostream: No such file or directory
 #include <iostream>
                    ^
compilation terminated.
make: *** [out/target/product/msm8916_64/obj_arm/SHARED_LIBRARIES/sensors.msm8916_intermediates/utils.o] Error 1
make: *** Waiting for unfinished jobs....
build/core/binary.mk:619: recipe for target 'out/target/product/msm8916_64/obj_arm/SHARED_LIBRARIES/sensors.msm8916_intermediates/utils.o' failed

mariotesi, 2019-09-11:
Hi, what version of Android are you building?

Tiny-Angel305, 2019-09-11:
Hi, I have already solved the issue myself, but thanks for your reply anyway.
Android version is Lollipop.

I have modified Android.mk as below;

LOCAL_SHARED_LIBRARIES += \
    libstlport

LOCAL_C_INCLUDES += $(LOCAL_PATH)/ \
			external/stlport/stlport \
			bionic \
			bionic/libstdc++/include

closed 2019-09-11 by Tiny-Angel305